boltgolt/howdy #1018: Fedora 42

mdx0111, 2025-05-16:
Followed the instructions but I'm unable to install howdy on Fedora 42.

Tried with both principis/howdy and principis/howdy-beta and I get the following error message:

**principis/howdy:**
Failed to resolve the transaction:
Problem: package howdy-2.6.1-10.fc42.x86_64 from copr:copr.fedorainfracloud.org:principis:howdy requires pam_python, but none of the providers can be installed
  - conflicting requests
  - nothing provides libpython2.7.so.1.0()(64bit) needed by pam_python-1.0.8-4.fc41.x86_64 from copr:copr.fedorainfracloud.org:principis:howdy
You can try to add to command line:
  --skip-broken to skip uninstallable packages

**principis/howdy-beta:**
Failed to resolve the transaction:
Problem: conflicting requests
  - nothing provides python3dist(ffmpeg-python) needed by howdy-3.0.0-5.20250220gitaef35b5.fc42.x86_64 from copr:copr.fedorainfracloud.org:principis:howdy-beta
You can try to add to command line:
  --skip-broken to skip uninstallable packages

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution: Fedora 42

Howdy version (`sudo howdy version`): NA

freedy237, 2025-05-19:
https://github.com/freedy237/howdy-fedora-install

gokul-k101, 2025-06-03:
`dnf install https://kojipkgs.fedoraproject.org/packages/python-ffmpeg-python/0.2.0/8.fc41/noarch/python3-ffmpeg-python-0.2.0-8.fc41.noarch.rpm`
Install this package and then follow the official guide to instal howdy beta again

freedy237, 2025-06-04:
Ok thanks

neonyirenda, 2025-07-20:
> > `dnf install https://kojipkgs.fedoraproject.org/packages/python-ffmpeg-python/0.2.0/8.fc41/noarch/python3-ffmpeg-python-0.2.0-8.fc41.noarch.rpm` Install this package and then follow the official guide to instal howdy beta again
> 
> This doesn't actually work. I still get exactly the same error :
> 
> ```
> Problem: conflicting requests
>   - nothing provides python3dist(ffmpeg-python) needed by howdy-3.0.0-6.20250624gitd3ab993.fc42.x86_64 from copr:copr.fedorainfracloud.org:principis:howdy-beta
> ```

It would be best if you include the output when you attempt to install `ffmpeg-python` and if it's successfull.