boltgolt/howdy #1021: Install fails

DavidRLTG, 2025-06-05:
```
sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libavcodec-dev libavformat-dev libavutil-dev libdc1394-dev libexif-dev
  libexif-doc libgdcm-dev libgphoto2-dev libimath-dev libopencv-calib3d-dev
  libopencv-contrib-dev libopencv-dev libopencv-dnn-dev
  libopencv-features2d-dev libopencv-flann-dev libopencv-highgui-dev
  libopencv-imgcodecs-dev libopencv-imgproc-dev libopencv-java
  libopencv-ml-dev libopencv-objdetect-dev libopencv-photo-dev
  libopencv-shape-dev libopencv-stitching-dev libopencv-superres-dev
  libopencv-superres406t64 libopencv-video-dev libopencv-videoio-dev
  libopencv-videostab-dev libopencv-videostab406t64 libopencv-viz-dev
  libopencv406-jni libopenexr-dev libpam-python libraw1394-dev
  libraw1394-tools libswresample-dev libswscale-dev opencv-data
Suggested packages:
  nvidia-cuda-dev opencv-doc libpam-python-doc libraw1394-doc
The following NEW packages will be installed:
  howdy libavcodec-dev libavformat-dev libavutil-dev libdc1394-dev libexif-dev
  libexif-doc libgdcm-dev libgphoto2-dev libimath-dev libopencv-calib3d-dev
  libopencv-contrib-dev libopencv-dev libopencv-dnn-dev
  libopencv-features2d-dev libopencv-flann-dev libopencv-highgui-dev
  libopencv-imgcodecs-dev libopencv-imgproc-dev libopencv-java
  libopencv-ml-dev libopencv-objdetect-dev libopencv-photo-dev
  libopencv-shape-dev libopencv-stitching-dev libopencv-superres-dev
  libopencv-superres406t64 libopencv-video-dev libopencv-videoio-dev
  libopencv-videostab-dev libopencv-videostab406t64 libopencv-viz-dev
  libopencv406-jni libopenexr-dev libpam-python libraw1394-dev
  libraw1394-tools libswresample-dev libswscale-dev opencv-data
0 upgraded, 40 newly installed, 0 to remove and 198 not upgraded.
Need to get 0 B/26,4 MB of archives.
After this operation, 141 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Extracting templates from packages: 100%
Selecting previously unselected package libpam-python.
(Reading database ... 579901 files and directories currently installed.)
Preparing to unpack .../00-libpam-python_1.1.0~git20220701.1d4e111-0.4ubuntu3_am
d64.deb ...
Unpacking libpam-python (1.1.0~git20220701.1d4e111-0.4ubuntu3) ...
Selecting previously unselected package libopencv-flann-dev:amd64.
Preparing to unpack .../01-libopencv-flann-dev_4.6.0+dfsg-13.1ubuntu1_amd64.deb 
...
Unpacking libopencv-flann-dev:amd64 (4.6.0+dfsg-13.1ubuntu1) ...
Selecting previously unselected package libavutil-dev:amd64.
Preparing to unpack .../02-libavutil-dev_7%3a6.1.1-3ubuntu5_amd64.deb ...
Unpacking libavutil-dev:amd64 (7:6.1.1-3ubuntu5) ...
Selecting previously unselected package libswresample-dev:amd64.
Preparing to unpack .../03-libswresample-dev_7%3a6.1.1-3ubuntu5_amd64.deb ...
Unpacking libswresample-dev:amd64 (7:6.1.1-3ubuntu5) ...
Selecting previously unselected package libavcodec-dev:amd64.
Preparing to unpack .../04-libavcodec-dev_7%3a6.1.1-3ubuntu5_amd64.deb ...
Unpacking libavcodec-dev:amd64 (7:6.1.1-3ubuntu5) ...
Selecting previously unselected package libavformat-dev:amd64.
Preparing to unpack .../05-libavformat-dev_7%3a6.1.1-3ubuntu5_amd64.deb ...
Unpacking libavformat-dev:amd64 (7:6.1.1-3ubuntu5) ...
Selecting previously unselected package libraw1394-dev:amd64.
Preparing to unpack .../06-libraw1394-dev_2.1.2-2build3_amd64.deb ...
Unpacking libraw1394-dev:amd64 (2.1.2-2build3) ...
Selecting previously unselected package libdc1394-dev:amd64.
Preparing to unpack .../07-libdc1394-dev_2.2.6-4build1_amd64.deb ...
Unpacking libdc1394-dev:amd64 (2.2.6-4build1) ...
Selecting previously unselected package libexif-dev:amd64.
Preparing to unpack .../08-libexif-dev_0.6.24-1build2_amd64.deb ...
Unpacking libexif-dev:amd64 (0.6.24-1build2) ...
Selecting previously unselected package libgphoto2-dev:amd64.
Preparing to unpack .../09-libgphoto2-dev_2.5.31-2.1build2_amd64.deb ...
Unpacking libgphoto2-dev:amd64 (2.5.31-2.1build2) ...
Selecting previously unselected package libgdcm-dev.
[... 199 lines cut ...]
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.12.0-4build2) ...
Processing triggers for libc-bin (2.39-0ubuntu8.4) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

``

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Linux Mint 22.1

Howdy version (`sudo howdy version`): did not install, used ppa.

matyasmatta, 2025-06-06:
Same issue, Ubuntu 25.04

Val-d-emar, 2025-06-16:
I have the same error at Debian 12 

```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation o
r OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

```

kissunyeason, 2025-06-28:
> Hey i was facing the same problem but here is the Solution
> 
> 💻 How to Install Howdy (Face Unlock) on Ubuntu 24.04 [Working Method]
> 
> ```
> ✅ Tested & Working | IR camera | Python-safe install
> ```
> 
> 🔧 1. Add the Patched Howdy PPA (UbuntuHandbook) sudo add-apt-repository ppa:ubuntuhandbook1/howdy sudo apt update
> 
> 📥 2. Install Howdy sudo apt install howdy
> 
> ⚙️ 3. Fix the Camera Not Found Issue If it says "camera not found", run: sudo howdy config ( this is a text file ) device_path = /dev/video1 ( from 1 to many check ) (Use v4l2-ctl --list-devices to confirm IR cam path)
> 
> 👤 4. Add Your Face sudo howdy add
> 
> 🔓 5. Optional: Enable Login-Screen Face Unlock sudo nano /etc/pam.d/gdm-password
> 
> Add this above [@include](https://github.com/include) common-auth: auth sufficient pam_python.so /lib/security/howdy/Howdy.py Then reboot: sudo reboot
> 
> I hope this helps you if you find any issues please comment i would love to be part of the community and help others!!

** works!!! **

afalout, 2025-06-28:
Note to people tempted to install random PPAs you know nothing about, recommended by people like abbuba that never ever posted or contributed anything before and just discovered ChatGPT ( ✅ 🔧 📥 ⚙️ 👤 🔓)

Don't.

That is, unless you enjoy random people around the word having access to your computer and your data. If you do, please ignore my comment. 

If you dont, I suggest reading the installation instructions and understanding how installing Python packages works on your distribution.

abbuba, 2025-06-29:
Hey Afalout! I am using Lenovo Yoga Pro 7, which has ir camera, and i wanted it to work for Linux, but I was running into an issue with Python 668. After spending over two hours, I solved it and wanted to share my solution to the problem. To make it better understandable, I used ChatGPT. 

I mean I should aim to do more contributions like you did ( 3), but I am trying to help unlike some people right afalout ?

regards 
Mr.honesty

LordKiRon, 2025-07-03:
Well had same issue on Ubuntu 25.
Ater reading a lot :) was able , as suggested to continue installation by commenting out "pip" lines from install script , then obviously nothing worked , so had to install bunch of libraries using "apt install" , however , last one , "dlib" did not had apt version , so at the end installed it off pip with "--break-package-system" which is bad, but made things working (after a lot of headache configuring logitech camera IR, missing libraries , GNOME keyring password and whole bunch of other stuff.
Surely not greatest "out of the box" experience :(.
I really hope devs will fix , although I understand its more like a feature request to redu stuff and not rely on pip ...

chocothegreat, 2025-07-25:
> Hey i was facing the same problem but here is the Solution
>
> 💻 How to Install Howdy (Face Unlock) on Ubuntu 24.04 [Working Method]
>
> ```
> ✅ Tested & Working | IR camera | Python-safe install
> ```
>
> 🔧 1. [...]

This is the exact answer I found manually. The guy that posted this fix on that site also troubleshoots in the comments.

icmp15, 2025-08-02:
exact same issue as the OP.

 dpkg-reonfigure howdy failed with the following error

/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
"davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

---------------------------

i purged the howdy package and then tried reinstalling (using synaptic package manager) and got the following errors:

Selecting previously unselected package howdy.
(Reading database ... 269193 files and directories currently installed.)
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Starting IR camera check...

Trying "USB2.0 FHD UVC WebCam: USB2.0 F"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: b

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.12.0-4build2) ...
Not building database; man-db/auto-update is not 'true'.
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
A package failed to install.  Trying to recover:
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"
Attempting installation of missing data files
Downloading 3 required data files...
dlib_face_recognition_resnet 100%[=============================================>]  20.44M  4.86MB/s    in 4.5s    
mmod_human_face_detector.dat 100%[=============================================>] 678.41K  1.56MB/s    in 0.4s    
shape_predictor_5_face_landm 100%[=============================================>]   5.44M  3.15MB/s    in 1.7s    
Unpacking...

closed 2026-11-16 by boltgolt