boltgolt/howdy #1040: Install failed on Latest Mint Cinnamon Fresh Install

mtarasuik, 2025-08-22:
Can't upgrade pip, install numpy etc. on externally managed environment (Mint Cinnamon 22.1 fresh install)

Mint installation instructions listed are now invalid.

~$ sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  nvidia-cuda-dev
The following NEW packages will be installed:
  howdy
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 0 B/38.9 kB of archives.
After this operation, 166 kB of additional disk space will be used.
Selecting previously unselected package howdy.
(Reading database ... 503583 files and directories currently installed.)
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Starting IR camera check...

Trying "Microsoft LifeCam Rear: Microso"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: b

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\
w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation 
or OS distribution provider. You can override this, at the risk of breaking your
 Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit
 status 1
Processing triggers for man-db (2.12.0-4build2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

alesisjoan, 2025-08-22:
@mtarasuik try to run this command:

`python3 -m pip install --upgrade numpy --verbose`

and check the response. Also make sure that python3 points to your working python3 version (mine is 3.11)

you can change by doing this:

`sudo update-alternatives --config python3`

rbasto1, 2025-09-01:
I can confirm that Ubuntu 24.04 also fails due to the attempt to update pip inside the install script. I had to comment that line from the installation script to be able to complete the dpkg run.

alesisjoan, 2025-09-01:
> I can confirm that Ubuntu 24.04 also fails due to the attempt to update pip inside the install script. I had to comment that line from the installation script to be able to complete the dpkg run.

I was able to make it run by monkey parching lines inside the source. However my terminal on Linux mint is failing I had to revert the changes.

In that case, howdy is running but doesn't override the login, so making sudo in a terminal still prompting for password.

DGxInfinitY, 2025-09-13:
Having the same problem on latest 24.04.

closed 2026-11-16 by boltgolt