boltgolt/howdy #1045: Can't install on Debian 13 (Python 3)

ringthebell02, 2025-09-21:
I am an absolute linux noob, so please don't be mad if I am doing something wrong. This is the result when I installed howdy via gdebi:

root@Gage-Linux:/home/gage# sudo gdebi '/home/gage/Downloads/howdy_2.6.1 (1).deb' 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reading state information... Done

Howdy: Windows Hello style authentication for Linux.
 Use your built-in IR emitters and camera in combination with face recognition
 to prove who you are.
Do you want to install the software package? [y/N]:y
(Reading database ... 178611 files and directories currently installed.)
Preparing to unpack .../Downloads/howdy_2.6.1 (1).deb ...
Backup of Howdy config file created in /tmp/howdy_config_backup_v2.6.1.ini
Unpacking howdy (2.6.1) over (2.6.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"
Processing triggers for man-db (2.13.1-1) ...

I don't really know why it didn't work. I got the facial recognition files downloaded for Python. Please help!

----

I've searched for similar issues already, and my issue has not been reported yet.

Debian 13

Last version

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against Howdy v2. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt

ringthebell02, 2026-11-16:
Ah, never expected a response here. Now using Kubuntu, so I'll have to give V3 a try!